riscv-software-src/homebrew-riscv #113: M1 Max: installation fails because of `error: expected unqualified-id`

serejke, 2023-10-26:
### Commands
```
brew tap riscv-software-src/riscv
brew install riscv-tools
```

### OS
> Darwin serejke-mac.lan 21.6.0 Darwin Kernel Version 21.6.0: Sat Jun 18 17:07:22 PDT 2022; root:xnu-8020.140.41~1/RELEASE_ARM64_T6000 arm64

### Error

```
==> make
Last 15 lines from /Users/serejke/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
In file included from /private/tmp/riscv-gnu-toolchain-20231026-4622-wdui2o/gdb/gdb/language.h:26:
/private/tmp/riscv-gnu-toolchain-20231026-4622-wdui2o/gdb/gdb/symtab.h:307:23: error: expected unqualified-id
        style = split_style::DOT;
                             ^
m2-exp.c:163:13: note: expanded from macro 'DOT'
#define DOT 302
            ^
  CXX    macrocmd.o
  CXX    macroexp.o
2 errors generated.
make[3]: *** [m2-exp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [all-gdb] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-gdb-newlib] Error 2
```

yulu83, 2023-11-02:
I had the same issue in the 07.make file

sbeamer, 2024-01-11:
This is due to bison on macOS being too old. It can be fixed with `brew install bison`.

However, since few folks use gdb in a cross-compiled environment, I disabled gdb support being on by default.

closed 2024-01-11 by sbeamer