riscv-software-src/homebrew-riscv #114: M2 Mac Air, riscv-gnu-toolchain install failed

abroim, 2023-11-02:
the error occurs when executing "make" after the commad.

Last 15 lines from /Users/chohyoungjune/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
	    if [ -f $i ]; then \
	     /usr/bin/install -c -m 644 $i /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/machine/`basename $i`; \
	    else true; fi ; \
	  done ; \
	  for i in rpc/types.h rpc/xdr.h; do \
	    if [ -f /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i ]; then \
		rm /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i; \
	    else true; fi ; \
	  done ; \
	else true; fi
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/rpc
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/ssp
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/sys
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/bits
mkdir -p stamps/ && touch stamps/build-newlib

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv/homebrew-riscv/issues

/opt/homebrew/Library/Homebrew/utils/github/api.rb:352:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
	from /opt/homebrew/Library/Homebrew/utils/github/api.rb:268:in `open_rest'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:178:in `search'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:182:in `search_results_items'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:36:in `search_issues'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:62:in `issues_for_formula'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:502:in `fetch_issues'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:497:in `issues'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:553:in `dump'
	from /opt/homebrew/Library/Homebrew/brew.rb:143:in `rescue in <main>'
	from /opt/homebrew/Library/Homebrew/brew.rb:131:in `<main>'
/opt/homebrew/Library/Homebrew/formula.rb:2737:in `block in system': Failed executing: make (BuildError)
	from /opt/homebrew/Library/Homebrew/formula.rb:2673:in `open'
	from /opt/homebrew/Library/Homebrew/formula.rb:2673:in `system'
	from /opt/homebrew/Library/Taps/riscv/homebrew-riscv/riscv-gnu-toolchain.rb:71:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:177:in `block (3 levels) in install'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:499:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:139:in `block (2 levels) in install'
	from /opt/homebrew/Library/Homebrew/formula.rb:1418:in `block in brew'
	from /opt/homebrew/Library/Homebrew/formula.rb:2904:in `block (2 levels) in stage'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:499:in `with_env'
	from /opt/homebrew/Library/Homebrew/formula.rb:2903:in `block in stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:129:in `block (2 levels) in unpack'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:121:in `chdir'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:113:in `stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:125:in `block in unpack'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `block in run'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `chdir'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `run'
	from /opt/homebrew/Library/Homebrew/resource.rb:226:in `stage_resource'
	from /opt/homebrew/Library/Homebrew/resource.rb:124:in `unpack'
	from /opt/homebrew/Library/Homebrew/resource.rb:98:in `stage'
	from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:2883:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:1411:in `brew'
	from /opt/homebrew/Library/Homebrew/build.rb:133:in `block in install'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:499:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:128:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:229:in `<main>'

maririn312, 2023-11-26:
is there anyone fixed this error

qkunzh, 2023-11-27:
i got the same error too

davidbroughsmyth, 2023-11-28:
Also same here. Have tried `brew install --cc=gcc-10 riscv-tools` and with gcc-14 with same errors.

==> Pouring readline--8.2.7.arm64_sonoma.bottle.tar.gz
🍺  /opt/homebrew/Cellar/readline/8.2.7: 50 files, 1.7MB
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-software-src/riscv/riscv-gnu-toolchain
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive binutils
==> git submodule update --init --recursive gcc
==> sed -i .bak s/.*=host-darwin.o$// gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// gcc/gcc/config.host
==> ./configure --with-cmodel=medany --enable-multilib
==> make
Last 15 lines from /Users/davidb/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
	    if [ -f $i ]; then \
	     /usr/bin/install -c -m 644 $i /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/machine/`basename $i`; \
	    else true; fi ; \
	  done ; \
	  for i in rpc/types.h rpc/xdr.h; do \
	    if [ -f /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i ]; then \
		rm /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i; \
	    else true; fi ; \
	  done ; \
	else true; fi
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/rpc
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/ssp
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/sys
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/bits
mkdir -p stamps/ && touch stamps/build-newlib

Do not report this issue to Homebrew/brew or Homebrew/homebrew-core!

These open issues may also help:
M2 Mac Air, riscv-gnu-toolchain install failed https://github.com/riscv-software-src/homebrew-riscv/issues/114

WOTMCCR, 2023-11-28:
also same here with command brew install --cc=gcc-10 riscv-tools   
Installing riscv-software-src/riscv/riscv-tools dependency: riscv-software-src/riscv/riscv-gnu-toolchain
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive binutils
==> git submodule update --init --recursive gcc
==> sed -i .bak s/.*=host-darwin.o$// gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// gcc/gcc/config.host
==> ./configure --with-cmodel=medany --enable-multilib
==> make
Last 15 lines from /Users/caichenru/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
Making install in testsuite
make[9]: Nothing to be done for `install-exec-am'.
make[9]: Nothing to be done for `install-data-am'.
Making install in python
make[9]: Nothing to be done for `install-exec-am'.
 ../../../../../.././gcc/libstdc++-v3/../install-sh -c -d '/opt/homebrew/Cellar/riscv-gnu-toolchain/main/share/gcc-13.2.0/python'
 /usr/bin/install -c -m 644 gdb.py /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/lib/rv64imafdc/lp64d/libstdc++.a-gdb.py
 ../../../../../.././gcc/libstdc++-v3/../install-sh -c -d '/opt/homebrew/Cellar/riscv-gnu-toolchain/main/share/gcc-13.2.0/python/libstdcxx/v6'
 /usr/bin/install -c -m 644  ../../../../../.././gcc/libstdc++-v3/python/libstdcxx/v6/printers.py ../../../../../.././gcc/libstdc++-v3/python/libstdcxx/v6/xmethods.py ../../../../../.././gcc/libstdc++-v3/python/libstdcxx/v6/__init__.py '/opt/homebrew/Cellar/riscv-gnu-toolchain/main/share/gcc-13.2.0/python/libstdcxx/v6'
 ../../../../../.././gcc/libstdc++-v3/../install-sh -c -d '/opt/homebrew/Cellar/riscv-gnu-toolchain/main/share/gcc-13.2.0/python/libstdcxx'
 /usr/bin/install -c -m 644  ../../../../../.././gcc/libstdc++-v3/python/libstdcxx/__init__.py '/opt/homebrew/Cellar/riscv-gnu-toolchain/main/share/gcc-13.2.0/python/libstdcxx'
true "AR_FLAGS=rc" "CC_FOR_BUILD=clang" "CC_FOR_TARGET=/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-newlib-stage2/./gcc/xgcc -B/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-newlib-stage2/./gcc/" "CFLAGS=-Os    -mcmodel=medany  -march=rv64imafdc -mabi=lp64d" "CXXFLAGS=-Os    -mcmodel=medany  -march=rv64imafdc -mabi=lp64d" "CFLAGS_FOR_BUILD=-g -O2" "CFLAGS_FOR_TARGET=-Os    -mcmodel=medany" "INSTALL=/usr/bin/install -c" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "INSTALL_SCRIPT=/usr/bin/install -c" "LDFLAGS=-march=rv64imafdc -mabi=lp64d" "LIBCFLAGS=-Os    -mcmodel=medany  -march=rv64imafdc -mabi=lp64d" "LIBCFLAGS_FOR_TARGET=-Os    -mcmodel=medany" "MAKE=/Applications/Xcode.app/Contents/Developer/usr/bin/make" "MAKEINFO=makeinfo --split-size=5000000 --split-size=5000000      " "SHELL=/bin/sh" "RUNTESTFLAGS=" "exec_prefix=/opt/homebrew/Cellar/riscv-gnu-toolchain/main" "infodir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/share/info" "libdir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib" "includedir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/include" "prefix=/opt/homebrew/Cellar/riscv-gnu-toolchain/main" "tooldir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf" "gxx_include_dir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/c++/13.2.0" "AR=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/bin/ar" "AS=/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-newlib-stage2/./gcc/as" "LD=/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-newlib-stage2/./gcc/collect-ld" "RANLIB=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/bin/ranlib" "NM=/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-newlib-stage2/./gcc/nm" "NM_FOR_BUILD=" "NM_FOR_TARGET=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/bin/nm" "DESTDIR=" "WERROR=" DO=all multi-do # /Applications/Xcode.app/Contents/Developer/usr/bin/make
true "AR_FLAGS=rc" "CC_FOR_BUILD=clang" "CC_FOR_TARGET=/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-newlib-stage2/./gcc/xgcc -B/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-newlib-stage2/./gcc/" "CFLAGS=-Os    -mcmodel=medany  -march=rv64imafdc -mabi=lp64d" "CXXFLAGS=-Os    -mcmodel=medany  -march=rv64imafdc -mabi=lp64d" "CFLAGS_FOR_BUILD=-g -O2" "CFLAGS_FOR_TARGET=-Os    -mcmodel=medany" "INSTALL=/usr/bin/install -c" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "INSTALL_SCRIPT=/usr/bin/install -c" "LDFLAGS=-march=rv64imafdc -mabi=lp64d" "LIBCFLAGS=-Os    -mcmodel=medany  -march=rv64imafdc -mabi=lp64d" "LIBCFLAGS_FOR_TARGET=-Os    -mcmodel=medany" "MAKE=/Applications/Xcode.app/Contents/Developer/usr/bin/make" "MAKEINFO=makeinfo --split-size=5000000 --split-size=5000000       " "SHELL=/bin/sh" "RUNTESTFLAGS=" "exec_prefix=/opt/homebrew/Cellar/riscv-gnu-toolchain/main" "infodir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/share/info" "libdir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib" "includedir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/include" "prefix=/opt/homebrew/Cellar/riscv-gnu-toolchain/main" "tooldir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf" "gxx_include_dir=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/c++/13.2.0" "AR=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/bin/ar" "AS=/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-newlib-stage2/./gcc/as" "LD=/private/tmp/riscv-gnu-toolchain-20231128-92692-h4nxfv/build-gcc-ne [... cut ...] RGET=/opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/bin/nm" "DESTDIR=" "WERROR=" DO=install multi-do # /Applications/Xcode.app/Contents/Developer/usr/bin/make
make[9]: Nothing to be done for `install-data-am'.
mkdir -p stamps/ && touch stamps/build-gcc-newlib-stage2

Do not report this issue to Homebrew/brew or Homebrew/homebrew-core!

These open issues may also help:
M2 Mac Air, riscv-gnu-toolchain install failed https://github.com/riscv-software-src/homebrew-riscv/issues/114

mtseng15, 2023-11-29:
I have also gotten the same error on OSX Sonoma 14.0. Is there any resolution?

maririn312, 2023-12-01:
> I have also gotten the same error on OSX Sonoma 14.0. Is there any resolution?

I searched it on internet i didn't get anything about it. I am waiting fix about this library so sad

ChelsyLan, 2023-12-06:
I got the same error on `mac pro m1 ventura 13.5`
`Last 15 lines from /Users/chengxulan/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
	    if [ -f /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i ]; then \
		rm /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i; \
	    else true; fi ; \
	  done ; \
	else true; fi
mkdir /private/tmp/riscv-gnu-toolchain-20231206-87587-mctrcy/install-newlib-nano/riscv64-unknown-elf/include/rpc
mkdir /private/tmp/riscv-gnu-toolchain-20231206-87587-mctrcy/install-newlib-nano/riscv64-unknown-elf/include/ssp
mkdir /private/tmp/riscv-gnu-toolchain-20231206-87587-mctrcy/install-newlib-nano/riscv64-unknown-elf/include/sys
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/rpc
mkdir /private/tmp/riscv-gnu-toolchain-20231206-87587-mctrcy/install-newlib-nano/riscv64-unknown-elf/include/bits
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/ssp
mkdir -p stamps/ && touch stamps/build-newlib-nano
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/sys
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/bits
mkdir -p stamps/ && touch stamps/build-newlib

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv-software-src/homebrew-riscv/issues

These open issues may also help:
M2 Mac Air, riscv-gnu-toolchain install failed https://github.com/riscv-software-src/homebrew-riscv/issues/114`

ChelsyLan, 2023-12-06:
I got the same error on `mac pro m1 ventura 13.5`
```
Last 15 lines from /Users/chengxulan/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
	    if [ -f /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i ]; then \
		rm /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/$i; \
	    else true; fi ; \
	  done ; \
	else true; fi
mkdir /private/tmp/riscv-gnu-toolchain-20231206-87587-mctrcy/install-newlib-nano/riscv64-unknown-elf/include/rpc
mkdir /private/tmp/riscv-gnu-toolchain-20231206-87587-mctrcy/install-newlib-nano/riscv64-unknown-elf/include/ssp
mkdir /private/tmp/riscv-gnu-toolchain-20231206-87587-mctrcy/install-newlib-nano/riscv64-unknown-elf/include/sys
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/rpc
mkdir /private/tmp/riscv-gnu-toolchain-20231206-87587-mctrcy/install-newlib-nano/riscv64-unknown-elf/include/bits
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/ssp
mkdir -p stamps/ && touch stamps/build-newlib-nano
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/sys
mkdir /opt/homebrew/Cellar/riscv-gnu-toolchain/main/riscv64-unknown-elf/include/bits
mkdir -p stamps/ && touch stamps/build-newlib

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv-software-src/homebrew-riscv/issues

These open issues may also help:
M2 Mac Air, riscv-gnu-toolchain install failed https://github.com/riscv-software-src/homebrew-riscv/issues/114
```

davidbroughsmyth, 2023-12-07:
As mentioned in issue#118 it now works for me. 
Not sure what magic has been done to make it work but I had to run `brew tap riscv-software-src/riscv` first followed by `brew install  riscv-tools`

abroim, 2023-12-07:
> As mentioned in issue#118 it now works for me. Not sure what magic has been done to make it work but I had to run `brew tap riscv-software-src/riscv` first followed by `brew install riscv-tools`

this work for me too, thx :D