riscv-software-src/homebrew-riscv #116: Failed to build on Monterey 12.6.9 on Intel during riscv-softw installation

scjody, 2023-11-07:
```
brew install riscv-tools
[...]
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-softw
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive binutils
==> git submodule update --init --recursive gcc
==> sed -i .bak s/.*=host-darwin.o$// gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// gcc/gcc/config.host
==> ./configure --with-cmodel=medany --enable-multilib
==> make
Last 15 lines from /Users/scjody/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
../.././gcc/gcc/config/riscv/genrvv-type-indexer.cc:120:30: error: no member named 'log2' in namespace 'std'; did you mean simply 'log2'?
    elmul_log2 = lmul_log2 + std::log2 (eew / sew);
                             ^~~~~~~~~
                             log2
/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/c++/v1/math.h:1463:1: note: 'log2' declared here
log2(_A1 __lcpp_x) _NOEXCEPT {return ::log2((double)__lcpp_x);}
^
2 errors generated.
make[2]: *** [build/genrvv-type-indexer.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rm fsf-funding.pod gpl.pod cpp.pod gcov-dump.pod gfdl.pod gcc.pod gcov.pod lto-dump.pod gcov-tool.pod
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2
make: *** Waiting for unfinished jobs....
Submodule path 'gdb': checked out '662243de0e14a4945555a480dca33c0e677976eb'
```

I don't see a repo or issue tracker for riscv-softw and the error message says to report issues here, but please redirect me if there's a better place for this.

AndrewTolmach, 2023-11-29:
I'm getting the same error on M1 running Monterey 12.7.1:
```
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-software-src/riscv/riscv-gnu-toolchain
==> git submodule update --init --recursive newlib
==> git submodule update --init --recursive binutils
==> git submodule update --init --recursive gcc
==> sed -i .bak s/.*=host-darwin.o$// gcc/gcc/config.host
==> sed -i .bak s/.* x-darwin.$// gcc/gcc/config.host
==> ./configure --with-cmodel=medany --enable-multilib
==> make
Last 15 lines from >
/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/c++/v1/math.h:1463:1: note: 'log2' declared here
log2(_A1 __lcpp_x) _NOEXCEPT {return ::log2((double)__lcpp_x);}
^
2 errors generated.
make[2]: *** [build/genrvv-type-indexer.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/bin/sh ../.././gcc/gcc/../move-if-change tmp-check.h tree-check.h
echo timestamp > s-check
/bin/sh ../.././gcc/gcc/../move-if-change tmp-mlib.h multilib.h
echo timestamp > s-mlib
rm fsf-funding.pod gpl.pod cpp.pod gcov-dump.pod gfdl.pod gcc.pod gcov.pod lto-dump.pod gcov-tool.pod
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2
make: *** Waiting for unfinished jobs....
Submodule path 'gdb': checked out '662243de0e14a4945555a480dca33c0e677976eb'
```
Any pointers greatly appreciated...

sbeamer, 2024-01-11:
I can't reproduce the error, but I do not have machine with macOS 12 currently. The error message appears to originating in rvv code, which in general, is quite recent. I suspect the issue might be that code is expecting a different C++ version. This error may not have occurred for others since their compiler's default C++ version is sufficient. The gcc code could be more compatible if it made the version expectation explicit with a compiler flag.

Thus, this might be an issue with riscv-gnu-toolchain.

sbeamer, 2024-01-11:
This is probably the same issue of #103. Since I already commented here, I am going to keep this one open and close the other.

lijqhs, 2024-04-19:
The same issue with Monterey 12.7.4

```
 ~  brew install riscv-tools
==> Fetching dependencies for riscv/riscv/riscv-tools: riscv/riscv/riscv-gnu-toolchain, riscv/riscv/riscv-isa-sim and riscv/riscv/riscv-pk
==> Fetching riscv/riscv/riscv-gnu-toolchain
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/user/Library/Caches/Homebrew/riscv-gnu-toolchain--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at f133b29 Merge pull request #1440 from ff520git/ff520git-patch-1
==> Fetching riscv/riscv/riscv-isa-sim
==> Cloning https://github.com/riscv/riscv-isa-sim.git
Updating /Users/user/Library/Caches/Homebrew/riscv-isa-sim--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 3192ee4d Merge pull request #1595 from Siudya/until-paddr
==> Fetching riscv/riscv/riscv-pk
==> Cloning https://github.com/riscv/riscv-pk.git
Updating /Users/user/Library/Caches/Homebrew/riscv-pk--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 6b5c8db mprotect whole ELF segment, not just file portion
==> Fetching riscv/riscv/riscv-tools
==> Downloading https://github.com/riscv/riscv-tools/archive/homebrew.tar.gz
Already downloaded: /Users/user/Library/Caches/Homebrew/downloads/5e080ac67877162e34a9e6701d88df637ec2f4abb27be9857918a142c6a3477e--riscv-tools-homebrew.tar.gz
==> Installing riscv-tools from riscv/riscv
==> Installing dependencies for riscv/riscv/riscv-tools: riscv/riscv/riscv-gnu-toolchain, riscv/riscv/riscv-isa-sim and riscv/riscv/riscv-pk
==> Installing riscv/riscv/riscv-tools dependency: riscv/riscv/riscv-gnu-to
==> git submodule update --depth 1 --init --recursive newlib
==> git submodule update --depth 1 --init --recursive binutils
==> git submodule update --depth 1 --init --recursive gcc
==> ./configure --with-cmodel=medany --disable-gdb --enable-multilib
==> make
Last 15 lines from /Users/user/Library/Logs/Homebrew/riscv-gnu-toolchain/05.make:
log2(_A1 __lcpp_x) _NOEXCEPT {return ::log2((double)__lcpp_x);}
^
../.././gcc/gcc/config/riscv/genrvv-type-indexer.cc:120:30: error: no member named 'log2' in namespace 'std'; did you mean simply 'log2'?
    elmul_log2 = lmul_log2 + std::log2 (eew / sew);
                             ^~~~~~~~~
                             log2
/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/c++/v1/math.h:1463:1: note: 'log2' declared here
log2(_A1 __lcpp_x) _NOEXCEPT {return ::log2((double)__lcpp_x);}
^
2 errors generated.
make[2]: *** [build/genrvv-type-indexer.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rm fsf-funding.pod gpl.pod cpp.pod gcov-dump.pod gfdl.pod gcc.pod gcov.pod lto-dump.pod gcov-tool.pod
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv/homebrew-riscv/issues

/usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:378:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
	from /usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:272:in `open_rest'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:178:in `search'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:182:in `search_results_items'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:36:in `search_issues'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:62:in `issues_for_formula'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:482:in `fetch_issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:477:in `issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:533:in `dump'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:149:in `rescue in <main>'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:137:in `<main>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:2833:in `block in system': Failed executing: make (BuildError)
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2769:in `open'
[... 2 lines cut ...]
	from /usr/local/Homebrew/Library/Homebrew/build.rb:179:in `block (3 levels) in install'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:495:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:139:in `block (2 levels) in install'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:1418:in `block in brew'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:3001:in `block (2 levels) in stage'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:495:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:3000:in `block in stage'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:129:in `block (2 levels) in unpack'
	from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:121:in `chdir'
	from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:113:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:125:in `block in unpack'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:75:in `block in run'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:75:in `chdir'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:75:in `run'
[... cut ...]
	from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/lib/ruby/3.1.0/forwardable.rb:238:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2980:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:1411:in `brew'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:133:in `block in install'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:495:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:125:in `install'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:231:in `<main>'
```

74HC14, 2024-08-13:
i also have this problem
`
../.././gcc/gcc/config/riscv/genrvv-type-indexer.cc:120:30: error: no member named 'log2' in namespace 'std'; did you mean simply 'log2'?
    elmul_log2 = lmul_log2 + std::log2 (eew / sew);
                             ^~~~~~~~~
                             log2
`
is there maybe a dirty fix by editing a file to change where log2 is referenced to? or by editing the file at /gcc/gcc/config/riscv/genrvv-etc, wherever that is. i'm probably going to have to upgrade my computer soon anyhow...

jreyess99, 2024-08-28:
I have the same problem, is there any solution for it?