riscv-software-src/homebrew-riscv #118: M1 Mac support

ColorlessBoy, 2023-12-06:
Hi, my system is `Apple M1 Pro-10151.41.12`.
When I run `brew install riscv-tools` I got error:

```bash
Last 15 lines from /Users/penglingwei/Library/Logs/Homebrew/riscv-gnu-toolchain/07.make:
        style = split_style::DOT;
                             ^
m2-exp.c:163:13: note: expanded from macro 'DOT'
#define DOT 302
            ^
  CXX    macrocmd.o
  CXX    macroexp.o
  CXX    macrotab.o
  CXX    macroscope.o
2 errors generated.
make[3]: *** [m2-exp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [all-gdb] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-gdb-newlib] Error 2
```

ykozhoma, 2023-12-06:
I have bumped into the same issue today on my M2 MBP (Version 14.1.2 (23B92))

ColorlessBoy, 2023-12-07:
You can try `brew install riscv-tools` again. It works on my computer now.

ykozhoma, 2023-12-09:
> You can try `brew install riscv-tools` again. It works on my computer now.

It worked for me! Huge thanks.

closed 2024-01-11 by sbeamer