riscv-software-src/homebrew-riscv #123: cant find gdb

ACCLE123, 2024-02-17:
I used
```
$ brew tap riscv/riscv
$ brew install riscv-tools
```
for downloading on arm64.
However,
```
(base) yangqi@yq ~ % riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (gc891d8dc2) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

(base) yangqi@yq ~ % riscv64-unknown-elf-gdb --version
zsh: command not found: riscv64-unknown-elf-gdb
```
And also, I couldn't find riscv64-unknown-elf-gdb in the bin directory.

bosscharlie, 2024-02-19:
Same problem.

sbeamer, 2024-02-19:
Correct. It does not build gdb.

closed 2024-03-17 by sbeamer