riscv-software-src/homebrew-riscv #126: Intel Mac: Install riscv-tools failed for Submodule newlib

Wei-Alwayers, 2024-04-20:
Device: Intel Mac
OS: Ventura 13.6.4

`brew install riscv-tools`

brew install riscv-tools Running `brew update --auto-update`...
==> Fetching dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Fetching riscv-software-src/riscv/riscv-gnu-toolchain
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
......
==> Installing riscv-tools from riscv-software-src/riscv
==> Installing dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-softw
==> git submodule update --depth 1 --init --recursive newlib

git
submodule
update
--depth
1
--init
--recursive
newlib

Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Cloning into '/private/tmp/riscv-gnu-toolchain-20240420-38095-wdj58y/newlib'...
error: Server does not allow request for unadvertised object 26f7004bf73c421c3fd5e5a6ccf470d05337b435
Fetched in submodule path 'newlib', but it did not contain 26f7004bf73c421c3fd5e5a6ccf470d05337b435. Direct fetching of that commit failed.

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv-software-src/homebrew-riscv/issues

escflee, 2024-04-23:
Same. Are there any workarounds for this problem?

chuckb, 2024-05-25:
Same here. I can do `git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain` and it clones (newlib included).

sbeamer, 2024-09-02:
I'm not able to reproduce the cloning issue, but it can be tricky to deal with these sorts of issues. Building from head appears stuck in general due to #127, so until that is fixed, it will be hard to confirm other build issues.

sbeamer, 2024-10-28:
Building is now fixed and I have pushed new pre-built bottles. Please let me know if there is any additional information.

closed 2025-08-13 by sbeamer