riscv-software-src/homebrew-riscv #127: MacOS Apple M2 Max, 14.4.1 (23E224), fails to build & install

zaun, 2024-05-07:
```
brew install riscv-tools                                                                                                     1 ↵
==> Fetching dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, dtc, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Fetching riscv-software-src/riscv/riscv-gnu-toolchain
Warning: Building riscv-software-src/riscv/riscv-gnu-toolchain from source as the bottle needs:
- HOMEBREW_CELLAR: /opt/homebrew/Cellar (yours is /Users/justinzaun/local/brew/Cellar)
- HOMEBREW_PREFIX: /opt/homebrew (yours is /Users/justinzaun/local/brew)
==> Cloning https://github.com/riscv/riscv-gnu-toolchain.git
Updating /Users/justinzaun/Library/Caches/Homebrew/riscv-gnu-toolchain--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 710a81b dejagnu: Bump version from c298959 to ca371cf
==> Fetching dtc
==> Downloading https://ghcr.io/v2/homebrew/core/dtc/manifests/1.7.0
Already downloaded: /Users/justinzaun/Library/Caches/Homebrew/downloads/00668efdbca5875f420621ff3011ae53448a6324636c2a27fe2323933c1ef3fa--dtc-1.7.0.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/dtc/blobs/sha256:a37a3658ee16c9b0428449ef970b771161f3f36b8f223276401277ebb582af57
Already downloaded: /Users/justinzaun/Library/Caches/Homebrew/downloads/fd52b018c55fb1f927d2fb58a19ca4bf295a782a143e90d7e2f748bc6261e862--dtc--1.7.0.arm64_sonoma.bottle.tar.gz
==> Fetching riscv-software-src/riscv/riscv-isa-sim
==> Downloading http://riscv.org.s3.amazonaws.com/bottles/riscv-isa-sim-main.arm64_sonoma.bottle.17.tar.gz
Already downloaded: /Users/justinzaun/Library/Caches/Homebrew/downloads/5fc6c5e498bfecab04254e38e32da258ad8ed03a52e14751d04d43df61fe888c--riscv-isa-sim-main.arm64_sonoma.bottle.17.tar.gz
==> Fetching riscv-software-src/riscv/riscv-pk
==> Downloading http://riscv.org.s3.amazonaws.com/bottles/riscv-pk-main.arm64_sonoma.bottle.15.tar.gz
Already downloaded: /Users/justinzaun/Library/Caches/Homebrew/downloads/273993ccca7d666dfe40cf8239548c30986cb9885c0871b6769d1d4994961b5b--riscv-pk-main.arm64_sonoma.bottle.15.tar.gz
==> Fetching riscv-software-src/riscv/riscv-tools
==> Downloading https://github.com/riscv/riscv-tools/archive/homebrew.tar.gz
Already downloaded: /Users/justinzaun/Library/Caches/Homebrew/downloads/5e080ac67877162e34a9e6701d88df637ec2f4abb27be9857918a142c6a3477e--riscv-tools-homebrew.tar.gz
==> Installing riscv-tools from riscv-software-src/riscv
==> Installing dependencies for riscv-software-src/riscv/riscv-tools: riscv-software-src/riscv/riscv-gnu-toolchain, dtc, riscv-software-src/riscv/riscv-isa-sim and riscv-software-src/riscv/riscv-pk
==> Installing riscv-software-src/riscv/riscv-tools dependency: riscv-software-src/riscv/riscv-gnu-toolchain
==> git submodule update --depth 1 --init --recursive newlib
==> git submodule update --depth 1 --init --recursive binutils
==> git submodule update --depth 1 --init --recursive gcc
==> ./configure --with-cmodel=medany --disable-gdb --enable-multilib
==> make
Last 15 lines from /Users/justinzaun/Library/Logs/Homebrew/riscv-gnu-toolchain/05.make:
    ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
clang++ -std=gnu++11  -fno-PIE -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o riscv-common.o -MT riscv-common.o -MMD -MP -MF ./.deps/riscv-common.TPo ../.././gcc/gcc/common/config/riscv/riscv-common.cc
clang++ -std=gnu++11  -fno-PIE -c  -DPREFIX=\"/Users/justinzaun/local/brew/Cellar/riscv-gnu-toolchain/main\" -DBASEVER="\"13.2.0\"" -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o prefix.o -MT prefix.o -MMD -MP -MF ./.deps/prefix.TPo ../.././gcc/gcc/prefix.cc
clang++ -std=gnu++11  -fno-PIE -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o opts.o -MT opts.o -MMD -MP -MF ./.deps/opts.TPo ../.././gcc/gcc/opts.cc
clang++ -std=gnu++11  -fno-PIE -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o opts-common.o -MT opts-common.o -MMD -MP -MF ./.deps/opts-common.TPo ../.././gcc/gcc/opts-common.cc
clang++ -std=gnu++11  -fno-PIE -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o options.o -MT options.o -MMD -MP -MF ./.deps/options.TPo options.cc
clang++ -std=gnu++11  -fno-PIE -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o vec.o -MT vec.o -MMD -MP -MF ./.deps/vec.TPo ../.././gcc/gcc/vec.cc
clang++ -std=gnu++11  -fno-PIE -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o hooks.o -MT hooks.o -MMD -MP -MF ./.deps/hooks.TPo ../.././gcc/gcc/hooks.cc
20 errors generated.
make[2]: *** [riscv-selftests.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rm fsf-funding.pod gpl.pod cpp.pod gcov-dump.pod gfdl.pod gcc.pod gcov.pod lto-dump.pod gcov-tool.pod
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2

Do not report this issue to Homebrew/brew or Homebrew/homebrew-core!

Do not report this issue: you are running in an unsupported configuration.
```

PratyushMakkar, 2024-05-23:
Were you able to find a workaround?

sbeamer, 2024-09-02:
I think this may be related to https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1536

We might be at the mercy of when riscv-gcc bumps to gcc 14.2.

sbeamer, 2024-10-28:
Building is now fixed and I have pushed new pre-built bottles. Please let me know if there is any additional information.