riscv-software-src/homebrew-riscv #129: M1 Macbook can not install riscv-openocd

janvier-vip, 2024-09-18:
```
> brew install riscv/riscv/riscv-openocd 
==> Fetching riscv/riscv/riscv-openocd
==> Cloning https://github.com/riscv/riscv-openocd.git
Updating /Users/janvier/Library/Caches/Homebrew/riscv-openocd--git
==> Checking out branch riscv
Already on 'riscv'
Your branch is up to date with 'origin/riscv'.
HEAD is now at 90162b325 Merge pull request #1130 from en-sc/en-sc/fix-reg-hide-warning
Entering 'jimtcl'
Entering 'src/jtag/drivers/libjaylink'
Entering 'tools/git2cl'
/Users/janvier/Library/Caches/Homebrew/riscv-openocd--git/jimtcl
/Users/janvier/Library/Caches/Homebrew/riscv-openocd--git/src/jtag/drivers/libjaylink
/Users/janvier/Library/Caches/Homebrew/riscv-openocd--git/tools/git2cl
==> Installing riscv-openocd from riscv/riscv
==> ./bootstrap nosubmodule
Last 15 lines from /Users/janvier/Library/Logs/Homebrew/riscv-openocd/01.bootstrap:
2024-09-18 03:33:24 +0000

./bootstrap
nosubmodule

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv/homebrew-riscv/issues

/opt/homebrew/Library/Homebrew/utils/github/api.rb:355:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
	from /opt/homebrew/Library/Homebrew/utils/github/api.rb:271:in `open_rest'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:178:in `search'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:182:in `search_results_items'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:36:in `search_issues'
	from /opt/homebrew/Library/Homebrew/utils/github.rb:62:in `issues_for_formula'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:502:in `fetch_issues'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/call_validation.rb:270:in `bind_call'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/call_validation.rb:270:in `validate_call'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/_methods.rb:272:in `block in _on_method_added'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:497:in `issues'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/call_validation.rb:270:in `bind_call'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/call_validation.rb:270:in `validate_call'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/_methods.rb:272:in `block in _on_method_added'
	from /opt/homebrew/Library/Homebrew/exceptions.rb:553:in `dump'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/call_validation.rb:270:in `bind_call'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/call_validation.rb:270:in `validate_call'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/_methods.rb:272:in `block in _on_method_added'
	from /opt/homebrew/Library/Homebrew/brew.rb:143:in `rescue in <main>'
	from /opt/homebrew/Library/Homebrew/brew.rb:131:in `<main>'
/opt/homebrew/Library/Homebrew/formula.rb:2744:in `block in system': Failed executing: ./bootstrap nosubmodule (BuildError)
	from /opt/homebrew/Library/Homebrew/formula.rb:2680:in `open'
	from /opt/homebrew/Library/Homebrew/formula.rb:2680:in `system'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/call_validation.rb:270:in `bind_call'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/call_validation.rb:270:in `validate_call'
	from /opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.1.0/gems/sorbet-runtime-0.5.11226/lib/types/private/methods/_methods.rb:272:in `block in _on_method_added'
	from /opt/homebrew/Library/Taps/riscv/homebrew-riscv/riscv-openocd.rb:23:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:177:in `block (3 levels) in install'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:493:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:139:in `block (2 levels) in install'
	from /opt/homebrew/Library/Homebrew/formula.rb:1412:in `block in brew'
	from /opt/homebrew/Library/Homebrew/formula.rb:2912:in `block (2 levels) in stage'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:493:in `with_env'
	from /opt/homebrew/Library/Homebrew/formula.rb:2911:in `block in stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:129:in `block (2 levels) in unpack'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:121:in `chdir'
	from /opt/homebrew/Library/Homebrew/download_strategy.rb:113:in `stage'
	from /opt/homebrew/Library/Homebrew/resource.rb:125:in `block in unpack'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `block in run'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `chdir'
	from /opt/homebrew/Library/Homebrew/mktemp.rb:75:in `run'
	from /opt/homebrew/Library/Homebrew/resource.rb:226:in `stage_resource'
	from /opt/homebrew/Library/Homebrew/resource.rb:124:in `unpack'
	from /opt/homebrew/Library/Homebrew/resource.rb:98:in `stage'
	from /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/lib/ruby/3.1.0/forwardable.rb:238:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:2891:in `stage'
	from /opt/homebrew/Library/Homebrew/formula.rb:1405:in `brew'
	from /opt/homebrew/Library/Homebrew/build.rb:133:in `block in install'
	from /opt/homebrew/Library/Homebrew/extend/kernel.rb:493:in `with_env'
	from /opt/homebrew/Library/Homebrew/build.rb:125:in `install'
	from /opt/homebrew/Library/Homebrew/build.rb:229:in `<main>'

```

sbeamer, 2024-10-28:
Any ideas @tuupola? Perhaps there is a new dependence that should be added?

tuupola, 2024-10-28:
I stopped using macs couple of years ago. Unable to help, sorry.

sbeamer, 2024-10-28:
@janvier-vip Can you try build openocd manually?
https://github.com/riscv-collab/riscv-openocd

Debugging within Homebrew is tricky. If you can build it without Homebrew, it usually isn't too hard to get it to build within Homebrew. A common cause for issues is the target project ends up requiring a new dependency.