riscv-software-src/homebrew-riscv #130: Make install fails on M2Pro, MacOS 13.7, CLT 15.1

luispimo, 2024-10-03:
```
==> make install
Last 15 lines from /Users/luis/Library/Logs/Homebrew/riscv-pk/02.make:
a - misaligned_ldst.o
a - misaligned_vec_ldst.o
a - flush_icache.o
a - mentry.o
a - fp_asm.o
riscv64-unknown-elf-ranlib libpk.a
riscv64-unknown-elf-ranlib libmachine.a
riscv64-unknown-elf-ranlib libsoftfloat.a
riscv64-unknown-elf-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static  -march=rv64imafdc_zifencei -mabi=lp64 -fno-stack-protector -o dummy_payload dummy_payload.o -L.  -ldummy_payload -lgcc -Wl,--defsym=MEM_START=0x80000000,-T,../dummy_payload/dummy_payload.lds
/opt/homebrew/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: dummy_payload has a LOAD segment with RWX permissions
../pk/pk.c: Assembler messages:
../pk/pk.c:122: Error: unknown CSR `ssp'
if riscv64-unknown-elf-readelf -h dummy_payload 2> /dev/null > /dev/null; then riscv64-unknown-elf-objcopy -O binary --set-section-flags .bss=alloc,load,contents dummy_payload bbl_payload; else cp dummy_payload bbl_payload; fi
make: *** [pk.o] Error 1
make: *** Waiting for unfinished jobs....
```

sbeamer, 2024-10-28:
With the fixes upstream and new bottles, this should hopefully all work. Please try installing with brew again.

luispimo, 2024-10-29:
Now it works! Thank you.

closed 2024-11-01 by sbeamer