riscv-software-src/homebrew-riscv #132: `brew cleanup` crashes due to this tap

wyan, 2025-02-25:
I'm running macOS 15.3 on an Apple Macbook Air M1, and normally `brew update && brew upgrade` several times a week.

Today I'm noticing that the cleanup step crashes with:

```
Error: Return value: Expected type String, got type BuildOptions with hash 4373652540369407951
Caller: /opt/homebrew/Library/Taps/riscv-software-src/homebrew-riscv/riscv-isa-sim.rb:17
Definition: /opt/homebrew/Library/Homebrew/formula.rb:3707 (Formula.build)
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/configuration.rb:296:in `call_validation_error_handler_default'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/configuration.rb:303:in `call_validation_error_handler'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:322:in `report_error'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation_2_7.rb:63:in `block in create_validator_method_fast0'
/opt/homebrew/Library/Taps/riscv-software-src/homebrew-riscv/riscv-isa-sim.rb:17:in `<class:RiscvIsaSim>'
/opt/homebrew/Library/Taps/riscv-software-src/homebrew-riscv/riscv-isa-sim.rb:1:in `block in load_formula'
/opt/homebrew/Library/Homebrew/formulary.rb:109:in `module_eval'
/opt/homebrew/Library/Homebrew/formulary.rb:109:in `block in load_formula'
/opt/homebrew/Library/Homebrew/formulary.rb:121:in `load_formula'
/opt/homebrew/Library/Homebrew/formulary.rb:152:in `load_formula_from_path'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:179:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:179:in `validate_call_skip_block_type'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:121:in `block in create_validator_slow_skip_block_type'
/opt/homebrew/Library/Homebrew/formulary.rb:528:in `load_file'
/opt/homebrew/Library/Homebrew/formulary.rb:734:in `load_file'
/opt/homebrew/Library/Homebrew/formulary.rb:519:in `klass'
/opt/homebrew/Library/Homebrew/formulary.rb:514:in `get_formula'
/opt/homebrew/Library/Homebrew/formulary.rb:724:in `get_formula'
/opt/homebrew/Library/Homebrew/formulary.rb:927:in `factory'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:179:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:179:in `validate_call_skip_block_type'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:121:in `block in create_validator_slow_skip_block_type'
/opt/homebrew/Library/Homebrew/dependency.rb:40:in `to_formula'
/opt/homebrew/Library/Homebrew/formula.rb:2400:in `block in runtime_formula_dependencies'
/opt/homebrew/Library/Homebrew/formula.rb:2399:in `each'
/opt/homebrew/Library/Homebrew/formula.rb:2399:in `filter_map'
/opt/homebrew/Library/Homebrew/formula.rb:2399:in `runtime_formula_dependencies'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:179:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:179:in `validate_call_skip_block_type'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:121:in `block in create_validator_slow_skip_block_type'
/opt/homebrew/Library/Homebrew/utils/autoremove.rb:40:in `block in bottled_formulae_with_no_formula_dependents'
/opt/homebrew/Library/Homebrew/utils/autoremove.rb:39:in `each'
/opt/homebrew/Library/Homebrew/utils/autoremove.rb:39:in `bottled_formulae_with_no_formula_dependents'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:282:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:282:in `validate_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/_methods.rb:277:in `block in _on_method_added'
/opt/homebrew/Library/Homebrew/utils/autoremove.rb:64:in `unused_formulae_with_no_formula_dependents'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:282:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:282:in `validate_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/_methods.rb:277:in `block in _on_method_added'
/opt/homebrew/Library/Homebrew/utils/autoremove.rb:16:in `removable_formulae'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:282:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:282:in `validate_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/_methods.rb:277:in `block in _on_method_added'
/opt/homebrew/Library/Homebrew/cleanup.rb:715:in `autoremove'
/opt/homebrew/Library/Homebrew/cleanup.rb:311:in `clean!'
/opt/homebrew/Library/Homebrew/cmd/cleanup.rb:52:in `run'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:282:in `bind_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/call_validation.rb:282:in `validate_call'
/opt/homebrew/Library/Homebrew/vendor/bundle/ruby/3.3.0/gems/sorbet-runtime-0.5.11851/lib/types/private/methods/_methods.rb:277:in `block in _on_method_added'
/opt/homebrew/Library/Homebrew/brew.rb:95:in `<main>'
If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/riscv-software-src/homebrew-riscv/issues/new
```

I can reproduce the crash with `brew cleanup`. Guessing a change in Homebrew's side, but can't investigate right now.

sbeamer, 2025-03-04:
Please share anything you can about this or how to reproduce.

I've been holding off on making new bottles and switching over to Sonoma because of the current issues in riscv-gnu-toolchain, but I do want to help with current problems.

sbeamer, 2025-08-13:
For arm64, we have switched to Sequoia. Is this issue still occurring?

wyan, 2025-08-13:
Sorry about the lack of answering, the issue went away by itself after some Homebrew update and is not happening any longer.